platformio/platform-nordicnrf51 #3: BBC MicroBit with S130 SoftDevice

wilstam, 2016-10-29:
_From @wilstam on October 27, 2016 3:39_

### Configuration

**Operating system**:
macOS Sierra 10.12.1

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.1.0

### Description of problem
There seems no setup for BBC MicroBit with S130 SoftDevice

#### Steps to Reproduce
1. Run 'pio run -v' on a MicroBit project. ie  [env:bbcmicrobit]
2. The linker script and the tool-sreccat will use S110 soft device by default.
3. There seems no way to config the build to use S130 soft device.
4. There has 'RF51_MICROBIT_B' setup under framework-mbed/hal/targets.json but there has no env has setup for it.

I have tried to change the following line to 'platforms/nordicnrf51/builder/frameworks/mbed/mbed.py:78

    "bbcmicrobit": "NRF51_MICROBIT_B"

The projecct is build with S130 and using correct linker script at 'packages/framework-mbed/hal/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_
ARM/TARGET_MCU_NRF51_16K_S130/NRF51822.ld'

Wonder what is the best way to add a support for this?

Thanks,
Wilson

_Copied from original issue: platformio/platformio#815_

wilstam, 2016-10-29:
_From @ivankravets on October 27, 2016 19:22_

Interesting question. Are these board the same? NRF51_MICROBIT & NRF51_MICROBIT_B?

wilstam, 2016-10-29:
Thanks Ivan.

I believe it has only one board so far. I bought two MicroBit boards six month apart, they should come from different production batches, both look the same and can run with S130 (ie NRF51_MICROBIT_B).

The closest thing I found is that few nRF51822 chips will not run on newer S130 version 2. As long as we use version 1, it will run on most chips.  The library in PIO right now is using version 1. 
[https://devzone.nordicsemi.com/question/93224/which-sdk11-gcc-examples-work-with-nrf51822/?comment=93336#comment-93336]
The two boards I have are using QFAAH0 chip and able to run version 1.

Looking at the _targets.json_, _NRF51_MICROBIT_ and _NRF51_MICROBIT_B_ share the same base _MCU_NRF51_16K_BASE_, only that _NRF51_MICROBIT_ has extra inherit from _MCU_NRF51_S110_ which specify the S110 SoftDevice. So I think it is safe to use _NRF51_MICROBIT_B_. 

S130 SoftDevice does use more space than S110 and provides BLE Central feature, which not many application will use, those may be the reason why the default is set to S110. As I am working on a project which requires BLE Central, I am trying to see if there any good or correct way to allow a project to specify which SoftDevice to be used.

Wonder, if it possible to set some variables in project's _platformio.ini_ to change the target setting? Or have to create a new board for this?

Cheers,
Wilson

wilstam, 2016-10-29:
_From @ivankravets on October 28, 2016 17:22_

We will create new board preset for MicrobitB. Please move this issue to this repo https://github.com/platformio/platform-nordicnrf51/issues

wilstam, 2016-10-29:
That's great. Thanks!

ivankravets, 2016-11-18:
Please use `board = bbcmicrobit_b`.

commit 4987f7e4 (2016-11-18): Add bbcmicrobit_b (NRF51_MICROBIT_B) target // Resolve #3
closed 2016-11-18 by ivankravets