platformio/platform-nordicnrf51 #6: Request for Simblee nrf51822 board support

danlsk, 2016-12-01:
Simblee Arduino library : [Board Manager URL](https://www.simblee.com/package_simblee166_index.json).
The library is not in github but you can get the files after installation in Arduino.

ivankravets, 2016-12-01:
Would you like to help us and provide support? It should be easy.
Here is docs http://docs.platformio.org/en/stable/platforms/creating_platform.html

danlsk, 2016-12-01:
Sure, I can try. In return, I get free upgrade? Joking.....

ivankravets, 2016-12-01:
> In return, I get free upgrade? Joking.....

Yes, you will receive free upgrade from PlatformIO 3.1 to PlatformIO 3.2 😂

simonem, 2017-04-05:
@danlsk did you manage to use simblee in platformio?

danlsk, 2017-04-07:
Nope, I didn't tried. I stop using PlatformIO at the moment.