platformio/platform-nordicnrf51 #8: BLE library is broken for nrf51xx

vkolotov, 2017-01-11:
A simple program fails to compile (Red Bear  BleNano):

`
#include "mbed.h"
#include "ble/BLE.h"
#include "DFUService.h"

void main() {
}
`

I've installed the following library:
pio lib install 296

And I'm getting the following errors:

`
[Thu Jan 12 08:23:46 2017] Processing redBearLabBLENano (platform: nordicnrf51, board: redBearLabBLENano, framework: mbed)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 8 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ble> v2.7.0

Compiling .pioenvs/redBearLabBLENano/src/main.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/BLE.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/BLEInstanceBase.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/DiscoveredCharacteristic.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/GapScanningParams.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/services/DFUService.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/services/UARTService.o
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/services/URIBeaconConfigService.o
src/main.cpp:3:24: fatal error: DFUService.h: No such file or directory
#include "DFUService.h"
^
compilation terminated.
In file included from .piolibdeps/ble_ID296/source/services/DFUService.cpp:19:0:
.piolibdeps/ble_ID296/ble/services/DFUService.h:26:29: fatal error: dfu_app_handler.h: No such file or directory
#include "dfu_app_handler.h"
^
compilation terminated.
*** [.pioenvs/redBearLabBLENano/src/main.o] Error 1
*** [.pioenvs/redBearLabBLENano/lib/ble_ID296/services/DFUService.o] Error 1
========================== [ERROR] Took 3.72 seconds ==========================

`

OS: MacOS Sierra 10.12.1
IDE: 1.7.1
Core: 3.2.1

matoushybl, 2017-01-12:
I can confirm this issue, same happens to me when building for Nordic DK.

wonderfulsuccess, 2017-01-14:
@matoushybl  Do you know how to solve this problem,  I mean how to use mbed BLE API in platformio.
Thanks a lot. 8 )

matoushybl, 2017-02-11:
I finally had some time to look into this problem and it seems that the build is broken for MacOS, I tried building on windows and there seemed to be no problem with it. Hopefully this could help a bit with looking for the problem.

pbochenski, 2017-02-20:
looks like .piolibdeps/nrf51-sdk_ID328/source/nordic_sdk/components/libraries/bootloader_dfu folder lacks of experimental sources. see mbed online compiler: http://imgur.com/a/73qkd
unfortunatelly coping those files there does not fix the problem.

surdu, 2017-06-14:
Any updates on this? It is till not working.

ivanmartinvalle, 2017-07-18:
If it helps anyone, it didn't work for me until I added `lib_deps = ble-nrf51822` into my platformio.ini file. I would have thought that `pio lib install ble-nrf51822` would do this for me, but it didn't.

ivankravets, 2018-01-24:
Fixed in the latest release

closed 2018-01-24 by ivankravets