platformio/platform-nordicnrf51 #10: Pinout config on bbc:microbit is wrong

Dealerpriest, 2017-02-20:
Hello! I'm using platformIO, IDE 1.7.2 | Core 3.2.1
I'm doing a project initialization as follows (using the microbit-hello-world example):
```
[env:bbcmicrobit]
platform = nordicnrf51
board = bbcmicrobit
framework = mbed
lib_deps =
   microbit@~2.0.0-rc4
```

The example code is this:
```
#define MICROBIT_DBG 1
#include <MicroBit.h>

MicroBit uBit;

int main() {
    // Initialise the micro:bit runtime.
    uBit.init();

    // Insert your code here!
    // while(1){
      uBit.display.scroll("HELLO WORLD! :)");
    // }

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
    return 0;
}
```
The example compiles fine and uploads to the microbit. But the pinout configuration seems to be somewhat off, as the led matrix doesn't display any sensible text. I've tried to use the equivalent example in the mbed-online-compiler, and the display scrolls HELLO WORLD as expected.

GilDev, 2017-09-07:
Same problem here. Any ideas?

josephroberts, 2017-09-08:
same problem here too.

tcurdt, 2017-12-09:
Anyone know where those pins are defined/mapped?

ugoebel, 2017-12-31:
Same problem. But I experimented a bit. I have exchanged the following two directories with those from the current branch of mbed:

<project>/.piolibdeps/microbit-dal_ID360/inc
<project>/.piolibdeps/microbit-dal_ID360/source

With this change the display works (buttons, motion sensor also). I have not tested everything. But it seems that the microbit-dal library is outdated.
I have a quick look on the differences on both branches. There are already some differences.
 
I am a completely new user of platformio. I think the concept is great. It was so easy to create an eclipse project for the "microbit" (I have not yet managed to do so under mbed).
It was just a pity that I immediately got this mistake. If anyone still knows how to update the microbit-dal library, that would be great.