platformio/platform-nordicnrf51 #11: nRF51822 and mbed lib in PlatformIO

pstyczynski, 2017-06-26:
Hello,
I'm developing code for board with nRF51822 chip. First I was writing it in mbed editor and everything was working fine.
Now I want to try PlatformIO integrated with VS Code. After installing everything, I created new project, added mbed and nrf51822 libs to it. Everything was fine, because simple blink code compile without error and after upload to board it works.

Then I wanted to test BLE and then I found problem. My code below:

```
#include "mbed.h"
#include "ble/BLE.h"
#include "BatteryService.h"

const static char DEVICE_NAME[] = "Button";
static const uint16_t uuid16_list[] = { GattService::UUID_BATTERY_SERVICE };

#define LED_BLINK_INTERVAL_S 0.1
#define LED_PIN P0_18
#define BUTTON_PIN P0_19

BLEDevice ble;
DigitalOut statusLed(LED_PIN);
Ticker statusLedTicker;
BatteryService *batteryServicePtr;

void statusLedCallback()
{
    statusLed = !statusLed;
}

void bleInitComplete(BLE::InitializationCompleteCallbackContext *params)
{
    BLE &ble          = params->ble;
    ble_error_t error = params->error;
 
    if (error != BLE_ERROR_NONE) 
    {
        return;
    }
 
    BatteryService batteryService(ble, false);
    batteryServicePtr = &batteryService;
 
    /* Setup advertising. */
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::BREDR_NOT_SUPPORTED | GapAdvertisingData::LE_GENERAL_DISCOVERABLE);
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LIST_16BIT_SERVICE_IDS, (uint8_t *)uuid16_list, sizeof(uuid16_list));
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LOCAL_NAME, (uint8_t *)DEVICE_NAME, sizeof(DEVICE_NAME));
    ble.gap().setAdvertisingType(GapAdvertisingParams::ADV_CONNECTABLE_UNDIRECTED);
    ble.gap().setAdvertisingInterval(1000); /* 1000ms */
    ble.gap().startAdvertising();
}

int main() 
{
    statusLedTicker.attach(statusLedCallback, LED_BLINK_INTERVAL_S);
    BLE& ble = BLE::Instance(BLE::DEFAULT_INSTANCE);
    ble.init(bleInitComplete);
    
    while (ble.hasInitialized()  == false) { /* spin loop */ }

    while (true) 
    { 
        ble.waitForEvent();
    }
}
```
Code compile without error, but after upload to device nothing happen. When I comment line:

`BLEDevice ble;`
led blink, but bleInitComplete don't run and I don't see my device on phone.
There is any way to debug step by step through constructor of BLEDevice to check what could happen?

zerog2k, 2017-07-29:
what board are you using?
most softdevices on mbed boards are expecting 32khz external crystal for LFOSC. Some generic modules may or may not have the extra LFOSC crystal, and need to be configured to use internal RC for LFOSC.

Also, I have had some issues, where the board I chose had mismatch with the board I was actually using in terms of ram. If this is wrong, I think the stack pointers will be initialized wrong, and wont work. (But I dont htink this is your problem, as led blink would work without initializing the ble class - so probably the previous issue I mentioned.)

Can you share your platformio.ini  and info about the actual board you are using?

ivankravets, 2018-01-24:
Could tou reproduce this issue with the latest dev/platform version?

closed 2018-02-02 by ivankravets