platformio/platform-nordicnrf51 #15: Compiled code with mbed for NRF51 doesn't run

Gecko05, 2017-09-04:
I'm relatively new to PlatformIO and I've been trying to run a simple example program for the NRF51. Had no trouble using mbed online with several examples but none of the examples in the PlatformIO seem to work on NRF51, can't even run a single blink led program. 
The only suspicious thing I get is "arm-none-eabi-g++: error: CreateProcess: No such file or directory" in my "main.cpp" file when saving the project.

valeros, 2017-09-05:
Hi, @Gecko05! What is your OS? Could you please provide full build and upload logs?

Gecko05, 2017-09-05:
I'm using Windows 10, here are my build logs [logs.txt](https://github.com/platformio/platform-nordicnrf51/files/1278319/logs.txt), unfortunately I lack the upload logs, I'm using a J-link to upload the hex generated from Platformio, I'm doing the same for the .hex generated from the mbed compiler. Hope it's enough, thanks.

valeros, 2017-09-06:
PlatformIO generates two `.hex` files:  `userfirmware.hex` and `firmware.hex`. Which one do you upload?

Gecko05, 2017-09-06:
firmware.hex

Gecko05, 2017-09-06:
Is there a way to upload the code via Platformio using the J-link?

zerog2k, 2018-01-20:
@Gecko05 nrfjprog

Gecko05, 2018-01-20:
Seems like my hardware is only able to run with a softdevice 130

closed 2018-01-20 by Gecko05