platformio/platform-nordicnrf51 #18: mbed-ble-thermometer example fails to link with redBearLabBLENano board

vshymanskyy, 2017-12-03:
```
...
Generating LD script .pioenvs/ble_nano/NRF51822.ld.link_script.ld
Linking .pioenvs/ble_nano/firmware.elf
.pioenvs/ble_nano/NRF51822.ld.link_script.ld:78 cannot move location counter backwards (from 0000000020004650 to 0000000020003800)
collect2: error: ld returned 1 exit status
*** [.pioenvs/ble_nano/firmware.elf] Error 1
```

The same example works with nrf51_dk.

ivankravets, 2018-01-24:
Fixed in the latest mbed OS.

vshymanskyy, 2018-01-26:
@ivankravets how do I use the the latest mbed OS?
I tried switching to development version of nordicnrf51 platform - still failing with same error.

vshymanskyy, 2018-01-26:
I use:
```
platform = https://github.com/platformio/platform-nordicnrf51.git
framework = mbed
board = redBearLabBLENano
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
upload_port = /media/${env.USER}/MBED/
```

ivankravets, 2018-01-26:
Which example do you use? I tested with `mbed-ble-thermometer` without `build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT`

vshymanskyy, 2018-01-26:
if I remove PIO_FRAMEWORK_MBED_RTOS_PRESENT, I can build the example! but then it doesn't work at all. Let me investigate if code executes at least...

vshymanskyy, 2018-01-27:
It looks code isn't even running... :(
```
[env:ble_nano]
platform = https://github.com/platformio/platform-nordicnrf51.git
framework = mbed
board = redBearLabBLENano
;build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
upload_port = /media/${env.USER}/MBED/
```
main.cpp:
```cpp
#include "mbed.h"

Serial pc(USBTX, USBRX);
DigitalOut led1(LED1, 1);

int main()
{
    while(1) {
      led1 = !led1;
      wait_ms(500);
      pc.printf("toggle\n");
    }

    return 0;
}
```
Run by:
```sh
pio run -e ble_nano --target upload && pio device monitor
```

vshymanskyy, 2018-01-27:
Also, when I specify 
```ini
board = nrf51_dk
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```
then it builds OK, but i can't test it on BLE Nano (different upload method)

vshymanskyy, 2018-01-27:
seeedTinyBLE - same issue.

michelle-russell, 2018-05-13:
Hi, any update on this bug? I am having similar issues on the seeedTinyBLE where a simple blinky/serial program appears to be compiling & uploading with no problems, yet no code runs. 

Some more info: An mbed online compiler hex output was 355KB and the platformio output from identical code was 288KB. I tried copying the mbed version into the .pioenvs/seedTinyBLE folder to see if platformio would upload that file, and it worked perfectly. Blinking and serial data galore. So, it seems that something isn't quite right in the build stage rather than the upload stage.

BleDudeAF, 2018-07-04:
any updates on this??

vshymanskyy, 2018-10-03:
@ivankravets still having issues with this. What nrf51-based device are you testing with?

danielkucera, 2018-10-03:
I have similar problem. This code compiled in April without problems:
https://github.com/danielkucera/ble-meter-mbed

Now I first had to add
```
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```

after that it throws:
```
Generating LD script .pioenvs/bbcmicrobit/NRF51822.ld.link_script.ld
Linking .pioenvs/bbcmicrobit/firmware.elf
.pioenvs/bbcmicrobit/NRF51822.ld.link_script.ld:78 cannot move location counter backwards (from 0000000020003f68 to 0000000020003800)
collect2: error: ld returned 1 exit status
*** [.pioenvs/bbcmicrobit/firmware.elf] Error 1
```

I luckily made a copy of the compiled version from April (.pioenvs) so I can provide it for investigation

ivankravets, 2018-10-12:
This board does not support RTOS. See https://os.mbed.com/platforms/RedBearLab-nRF51822/ (right column). Only SDK/MBED 2

vshymanskyy, 2018-10-13:
@ivankravets I think it is a bug. See this comment: https://github.com/platformio/platform-nordicnrf51/issues/18#issuecomment-360974747
The code isn't running even without RTOS enabled

danielkucera, 2019-05-27:
If someone is interrested, I was able to build my code with latest versions.
I have removed
```
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```
because my board doesn't support mbed-os (mbed 5) and copied BLE library which was required for my code by copying it from mbed `features`
More info #29

ivankravets, 2019-05-27:
@danielkucera what is your final `platformio.ini`?

danielkucera, 2019-05-27:
@ivankravets you can see it here:
https://github.com/danielkucera/ble-meter-mbed/blob/master/platformio.ini

vshymanskyy, 2019-06-17:
@ivankravets 
I was able to build and run PIOTherm (mbed OS 5) example with my `RedBearLab BLE Nano 1.5`.
But I had to use `nrf51_dk` target.

```
[env:nrf51_dk]
platform = nordicnrf51
framework = mbed
board = nrf51_dk
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT

board_build.mcu = nrf51822
board_build.f_cpu = 32000000L
upload_protocol = cmsis-dap
```

vshymanskyy, 2019-06-17:
It looks like `mbed OS 5` support should be enabled to all nrf51 variants with SRAM >= 32Kb