platformio/platform-nordicnrf51 #21: Device not running

daniel-maiorano, 2018-07-11:
Hi, I developed an app on MBED that runs on this kind of device:
http://www.wireless-tag.com/wireless_module/BLE/WT51822-S4AT.html

Even if MBED doesn't have a specific board for it, I use a RedBearLab BLE Nano board, and everything works fine.

I tried to port the project to Platformio but no way to make it work.
It compiles fine, it downloads fine, but it doesn't either run.

I suspected the size of the RAM (since the above mentioned board has only 16K), so I created a custom board following this: 
http://docs.platformio.org/en/latest/platforms/creating_board.html

but no luck
Any help will be appreciated

vshymanskyy, 2018-07-16:
I think this is duplicate of https://github.com/platformio/platform-nordicnrf51/issues/18

ivankravets, 2018-10-13:
Duplicate of https://github.com/platformio/platform-nordicnrf51/issues/18

closed 2018-10-13 by ivankravets