platformio/platform-nordicnrf51 #23: update arduino framework

zerog2k, 2018-08-06:
https://github.com/sandeepmistry/arduino-nRF5/releases
has 0.5.1 release which fixes some i2c issues ... current platform only has 0.4.0

ivankravets, 2018-08-08:
Thanks! Resolved in https://github.com/platformio/platform-nordicnrf51/commit/56c86da68f6c2f75f7db2f330236bc86e9f99c95

The final release is coming soon.

closed 2018-08-08 by ivankravets

zerog2k, 2018-08-10:
@ivankravets something doesnt look right with the updated framework-arduinonordicnrf5
http://dl.platformio.org/packages/framework-arduinonordicnrf5-0.5.1.tar.gz
doesnt match the contents of 0.5.1 tag on the target repo
https://github.com/sandeepmistry/arduino-nRF5/archive/0.5.1.tar.gz

in fact, in the framework package, the libraries and cores folders look old (have 26 Nov 17 date intead of 24 Feb 18)

How are these packages produced? It seems to be a bit of a mysterious black box.

ivankravets, 2018-08-11:
Sorry, fixed! Please `$ pio update`.