platformio/platform-nordicnrf51 #24: support for nordic sdk as framework

zerog2k, 2018-10-12:
We have support for arduino and mbed frameworks, but would like to investigate support for using nordics nrf5 sdk (hals/drivers, etc).

I think I understand about 80% of what needs to be done, but may need some help on polishing, patching, and packaging. 

Curious how many folks would be interested in such an effort?

Related question would be if it is possible for a user/board to specify a certain revision of the framework (sdk). Reason being is that nrf51 devices support maximum of nrf5 sdk 12.3.0, whereas nrf52 support latest versions. (nrf5 sdk version also dictates which softdevice is used, if ble softdevice is required, but i think this can be tied to and encapsulated into the framework logic?)

vshymanskyy, 2018-10-13:
+

shantiscript, 2018-12-21:
+1, having the Nordic SDK as a platform would be great

ASL07, 2018-12-26:
Very interesting stuff, let me know if I can help

tabchas, 2019-01-09:
Is work being started on this? Very interested in using native NRF sdk on PIO.

baneageorge, 2019-02-12:
@zerog2k Hi Jens, any chance we get in touch?

cortesis, 2019-02-18:
+1, would be very nice

andersliland, 2019-04-09:
+1, would be very nice

superhappychris, 2019-07-20:
+1 yes please!

hpssjellis, 2019-12-10:
Can someone comment on any progress that has been done to implement the Nordic SDK. I have an online docker that has all three of: Platformio, Nordic SDK and Openthread, working individually but have no idea how to integrate the three. Any suggestions would be appreciated.

The github is here [my-gitpod-of-openthread](https://github.com/hpssjellis/my-gitpod-of-openthread) 

Or just click on the button below to load the Gitpod docker (Needs a Github or Gitlab login)

[image]

Openthread can generate a .hex file for the  Nordic nRF52840 USB Dongle to be installed using nrf-connect for desktop

The Nordic SDK can generate .bin or .hex files for installing either using the Arduino IDE (with a bit of command line hack) or nrf-connect for desktop.

PlatformIO Client is installed and works great when connected with a local machine remote agent.

zerog2k, 2019-12-10:
I'm not aware of anyone actively working on this. I was trying to gauge interest and find others who could help support such a thing, but I have not really done much on nrf51 in a while - seems like most have moved to nrf52+ (although there are quite a few limited devices out there in the wild still using nrf51)

knud, 2020-01-12:
I've been doing both nRF51 and nRF52 work with the latest SDKs that support either, so 12.3 and 16, respectively, as of today. I would really like to get away from Eclipse as the environment and am considering PlatformIO. Would be willing to help since I might have learned useful things just getting the nRF51 stuff going (which was for a RedBear Nano v1, nRF51822).

hpssjellis, 2020-01-12:
@knud please look [here](https://community.platformio.org/t/like-if-you-have-an-nrf52840-nordic-usb-dongle-arduino-nano-33-ble-adafruit-sparkfun-april-makerdairy-fanstel-particle-or-any-other/11112/10) on platformIO community

I am having some success with various nRF52840 boards 
But having issues reassigning pins that aren't pre-defined for the Arduino 33 BLE. Any pin assigned I can reassign just confused if a nRF52840 SOC pin that hasn't been used by Arduino but is used by one of the other boards.

knud, 2020-01-12:
@hpssjellis Thanks for the quick ping back.

My interest is not only various dev kit/eval boards, but custom boards. I have several I’ve developed and the obvious, first step is to do a custom pin mapping thing. With the SDK that’s pretty straightforward. 

I’ll have to dig into how the boards supported for nRF52 on PIO are specified before I can be of much use. Coming late to the party, so to speak... and have not yet done too much with PIO...

hpssjellis, 2020-01-13:
@knud I made a pin map spreadsheet for the Nano33Ble, Nordic and makerdairy nrf52840 dongles and the particle xenon. It will have a few errors but should be a starting point. 
I just need to know how to get one undefined pin working and should be fine from there.

[https://github.com/hpssjellis/everything-nrf52840-usb-dongles/blob/master/images/general-pins03.ods](https://github.com/hpssjellis/everything-nrf52840-usb-dongles/blob/master/images/general-pins03.ods)

zerog2k, 2020-01-13:
I think I had documented or summarized my thoughts on how to setup a new sdk for an existing platform somewhere a year or so ago, let me see if I can dig it up. Unless perhaps someone has created a better tutorial. (I think this was why I was initially asking if anyone was already working on this, or had a shared group of interest who could help with parts of this.)

edit, found it:
https://community.platformio.org/t/custom-framework-development-guidance/4716/2

zerog2k, 2020-01-13:
I'm recollecting that one of the hurdles might have been how to (re-)distribute the nordic sdk automatically from platformio sdk/framework builds. If nothing else, I'm guessing it should be somehow possible to have a scripted download helper, or at the worst, require user to download and deploy the sdk into some location. Perhaps someone here has experience or knowledge about restributing the nordic nrf5* sdks?
Possibly related is the handling of softdevices. 
However, I'm sure that we could collectively come up with a creative solution that satisfies all of this.

(as a side note, I think even the nrf5 sdk 12.3.0 for example will need some patches, I myself had found a bug [typo in one of the proprietary radio functions] or maybe two in the sdk, and nordic basically said they would not be releasing even fixes to old sdks).

hpssjellis, 2020-01-13:
Glad both @knud and @zerog2k are interested in this. My approach is probably going to be to get as many nRF52840 boards working as I can using the Arduino IDE and the Nano 33 BLE bootloader, which does work very well PlatformIO. Then write code to activate the dead pins on the other boards. How to get these other boards working on PlatformIO I might leave to the experts as I am still fairly new to PlatformIO.

adrianschlatter, 2020-06-20:
+1 from my side.

Zhurakovsky, 2020-10-13:
+++

kb-consulting, 2020-12-04:
Yes please!

pingdan32, 2021-02-07:
++++++

MakeSenseIoT, 2021-02-14:
Yes please!

andy-kogsys, 2021-09-13:
Yes please!

TauqirH, 2021-12-27:
wen sdk support

TauqirH, 2022-03-29:
+100