platformio/platform-nordicnrf51 #27: No upload error, but uploading failed on RedBear BLE Nano (mbed)

baobabx, 2019-02-20:
I build and upload the "mbed-blink" example to my RedBear BLE Nano v1.5. (https://github.com/platformio/platform-nordicnrf51/tree/master/examples/mbed-blink)
There are no error as follows.
But It looks BLE Nano doesn't run new program.
What the matter ?

PlatformIO: 3.6.4
Nordic nRF51: 4.0.0

Processing redBearLabBLENano (platform: nordicnrf51; board: redBearLabBLENano; framework: mbed)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf51/redBearLabBLENano.html
PLATFORM: Nordic nRF51 > RedBearLab BLE Nano 1.5
HARDWARE: NRF51822 16MHz 32KB RAM (256KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 4 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\redBearLabBLENano\src\main.o
Linking .pioenvs\redBearLabBLENano\firmware.elf
Checking size .pioenvs\redBearLabBLENano\firmware.elf
Building .pioenvs\redBearLabBLENano\userfirmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   2.9% (used 960 bytes from 32768 bytes)
PROGRAM: [          ]   4.7% (used 12240 bytes from 262144 bytes)
merge_firmwares([".pioenvs\redBearLabBLENano\firmware.hex"], [".pioenvs\redBearLabBLENano\userfirmware.hex"])
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, nrfjprog, stlink
CURRENT: upload_protocol = mbed
Looking for upload disk...
Auto-detected: E:\
Uploading .pioenvs\redBearLabBLENano\firmware.hex
Firmware has been successfully uploaded.
(Some boards may require manual hard reset)

ivankravets, 2019-03-11:
Does this board has mbed bootloader(disk)?

neu-rah, 2019-09-07:
had similar issue in two cases:

1) with arduino framework because the softdevice file was missing, upload **only shows a warning about not finding the file** (in verbose mode) and terminates without error (firmware.hex contains only user version and is quite small)

2) with mbed OS 5, compile upload ok but nothing happens... by turning on Serial printing I was then able to see the error on terminal (serial monitor)

```text
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x1F351
Error Value: 0x0
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
-- MbedOS Error Info --
Please provide an implementation for mbed BLE
```

custom nRF51822xxaa board, correctly identified on build:  

```text
PLATFORM: Nordic nRF51 4.3.0 > ...
HARDWARE: NRF51822 16MHz, 16KB RAM, 256KB Flash`
```