platformio/platform-nordicnrf51 #29: Enabling feature

danielkucera, 2019-05-22:
I want to use mbed (2) without rtos but I'm getting following error:
```Processing bbcmicrobit (platform: nordicnrf51; board: bbcmicrobit; framework: mbed)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf51/bbcmicrobit.html
PLATFORM: Nordic nRF51 > BBC micro:bit
HARDWARE: NRF51822 16MHz 16KB RAM (256KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 4 compatible libraries
Scanning dependencies...
No dependencies
Warning! Unknown upload protocol stlink
Compiling .pioenvs/bbcmicrobit/src/main.o
src/main.cpp:20:10: fatal error: ble/BLE.h: No such file or directory
#include "ble/BLE.h"
^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/bbcmicrobit/src/main.o] Error 1
```

but the header should be available as a "feature":
```
./packages/framework-mbed/features/FEATURE_BLE/ble/BLE.h
```

How do I enable it?

danielkucera, 2019-05-22:
Ok, it compiles after:
```
cp -r ./packages/framework-mbed/features/FEATURE_BLE ~/projectdir/lib/
```
Would it be possible to make it available from the mbed core itself?

danielkucera, 2019-05-22:
It also works fine with `nordicnrf51 @ 3.5.0`
```
Processing core51822 (platform: nordicnrf51@3.5.0; board: bbcmicrobit; framework: mbed)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing nordicnrf51 @ 3.5.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
nordicnrf51 @ 3.5.0 has been successfully installed!
The platform 'nordicnrf51@3.5.0' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-mbed @ ~5.51001.1
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf51/bbcmicrobit.html
PLATFORM: Nordic nRF51 > BBC micro:bit
HARDWARE: NRF51822 16MHz 16KB RAM (256KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <mbed-events>
|-- <mbed-FEATURE_BLE>
Warning! Unknown upload protocol stlink
Compiling .pioenvs/core51822/FrameworkMbedCore/cmsis/TARGET_CORTEX_M/mbed_tz_context.o
Compiling .pioenvs/core51822/FrameworkMbedCore/drivers/AnalogIn.o
Compiling .pioenvs/core51822/FrameworkMbedCore/drivers/BusIn.o```

danielkucera, 2019-05-22:
Here is the project in case you want to test:
https://github.com/danielkucera/ble-meter-mbed

valeros, 2019-05-22:
Hi @danielkucera ! Thanks for the report. Are you able to compile your project using the online mbed compiler or mbed-cli tool?

danielkucera, 2019-05-22:
ok, it is probably not related to the platform but to `framework-mbed`
version `~5.51001.1` works but `~5.51105.190304` doesn't
```
    "framework-mbed": {
      "type": "framework",
      "optional": true,
      "versiona": "~5.51105.190304",
      "version": "~5.51001.1"
    },
```

danielkucera, 2019-05-22:
It has stopped working between these two versions:
```
5.51103.190208
5.51001.181029
```

danielkucera, 2019-05-23:
I dissected it to this code. The previous version loaded much more libs specified in board json file:
```
def get_mbed_config(target):
    config_file = join(FRAMEWORK_DIR, "platformio", "variants", target,
                       target + ".json")
    if not isfile(config_file):
        sys.stderr.write("Cannot find the configuration file for your board! "
                         "Please read instructions here %s\n" % join(
                             FRAMEWORK_DIR, "platformio", "README.txt"))
        env.Exit(1)

    return util.load_json(config_file)

# There is no difference in processing between lib and feature
libs = mbed_config.get("libs").copy()
libs.update(mbed_config.get("components"))
libs.update(mbed_config.get("features"))
libs.update(mbed_config.get("frameworks"))

for lib, lib_config in libs.items():
    if not lib_config:
        continue
    extra_includes = []
    env.Append(EXTRA_LIB_BUILDERS=[
        CustomLibBuilder(env, join(FRAMEWORK_DIR, lib_config.get("dir")),
                         get_dynamic_manifest(lib, lib_config, extra_includes))
    ])
```
the new one loads only a handful of them for non-RTOS mbed:
```
#
# Add legacy libs as standalone
#

if not MBED_RTOS:
    legacy_libs = (
        join(FRAMEWORK_DIR, "features", "unsupported", "dsp"),
        join(FRAMEWORK_DIR, "features", "unsupported", "rpc"),
        join(FRAMEWORK_DIR, "features", "unsupported", "USBDevice"),
        join(FRAMEWORK_DIR, "features", "unsupported", "USBHost")
    )

    for lib in legacy_libs:
        env.Append(EXTRA_LIB_BUILDERS=[
            PlatformIOLibBuilder(
                env, join(FRAMEWORK_DIR, lib),
                get_dynamic_manifest(lib))
        ])
```

valeros, 2019-05-23:
@danielkucera Have you tried compiling your project using the mbed online compiler?

danielkucera, 2019-05-23:
@valeros, I see your point but I don't know how to setup the online compiler correctly. I've never used it.

valeros, 2019-05-23:
I think it's reasonable to test your example using the official mbed tools. If I'm not mistaken, the BLE feature requires some RTOS functionality. As you can see in the mbed build script, `features` folder is ignored in mbed-2 mode. So your way of adding the BLE feature as a standalone library is probably the best workaround at the moment.

danielkucera, 2019-05-24:
Ok, I agree, I'll use it as a standalone library. Thanks for help.

closed 2019-05-24 by danielkucera