platformio/platform-nordicnrf51 #31: Firmware upload via st-link broken

amurchick, 2020-01-02:
Platformio version: 4.1.0 (mac os)

Platform Nordic nRF51:
```
Updating nordicnrf51                     @ 5.0.0          [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
Updating framework-arduinonordicnrf5     @ 1.600.190830   [Up-to-date]
Updating framework-mbed                  @ 5.51105.190312 [Up-to-date]
Updating tool-sreccat                    @ 1.164.0        [Up-to-date]
Updating tool-openocd                    @ 2.1000.190707  [Up-to-date]
Updating tool-nrfjprog                   @ 1.90702.1      [Up-to-date]`

```

```
$ pio run -v -t upload
[compiling messages skipped]
CURRENT: upload_protocol = stlink
openocd -d2 -f scripts/interface/stlink.cfg -c "transport select hla_swd; set WORKAREASIZE 0x4000" -f scripts/target/nrf51.cfg -c "program {.pio/build/redBearLab/firmware.hex}  verify reset; shutdown;"
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-15:21)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

embedded:startup.tcl:26: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
*** [upload] Error 1
```

When run openocd via command line - all OK (and firmware worked):
```
$ ~/.platformio/packages/tool-openocd/bin/openocd -s ~/.platformio/packages/tool-openocd -f scripts/interface/stlink.cfg -c "transport select hla_swd; set WORKAREASIZE 0x4000" -f scripts/target/nrf51.cfg -c "program {.pio/build/redBearLab/firmware.hex}  verify reset; shutdown"
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-15:21)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.273594
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
** Programming Started **
Info : nRF51822-QFAA(build code: G3) 256kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg/nrf52.cfg to disable it
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

platformio.ini
```
[env:redBearLab]
platform = nordicnrf51
board = redBearLab
board_build.variant = RedBearLab_nRF51822_devboard
framework = arduino

upload_protocol = stlink
debug_tool = stlink

build_flags =
  -Wl,-u,_printf_float,-u,_scanf_float
  -DNRF51_S130

monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200

lib_deps =
  # pio lib install "BLEPeripheral"
  259
```

valeros, 2020-01-02:
Hi @amurchick ! Thanks for the report. Please switch to the development version of the platform until a bugfix is released.
```
[env:redBearLab]
platform = https://github.com/platformio/platform-nordicnrf51.git
board = redBearLab
board_build.variant = RedBearLab_nRF51822_devboard
framework = arduino

upload_protocol = stlink
debug_tool = stlink

build_flags =
  -Wl,-u,_printf_float,-u,_scanf_float
  -DNRF51_S130

monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200

lib_deps =
  # pio lib install "BLEPeripheral"
  259
```

closed 2020-01-02 by valeros

amurchick, 2020-01-03:
It worked after
```
platform = https://github.com/platformio/platform-nordicnrf51.git
```

Thank you!

commit 02cc816b (2020-01-02): Fix OpenOCD server arguments // Resolve #31