platformio/platform-nordicnrf51 #32: TypeError: string indices must be integers:

hansj66, 2020-02-17:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

OS Name:                   Microsoft Windows 10 Home
OS Version:                10.0.18362 N/A Build 18362

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 4.2.0

### Description of problem
Build fails with "TypeError: string indices must be integers:" during dependency scan

#### Steps to Reproduce

1. create a new project. (BBC Microbit board, MBed framework)
2. add "lib_deps = microbit-dal" to platformio.ini
3. add "#include "MicroBitDisplay.h" to main.cpp (or any other microbit-dal header)

### Actual Results
> Executing task: C:\Users\hansj\.platformio\penv\Scripts\platformio.exe run <

Processing bbcmicrobit (platform: nordicnrf51; board: bbcmicrobit; framework: mbed)
----------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf51/bbcmicrobit.html
PLATFORM: Nordic nRF51 5.0.2 > BBC micro:bit
HARDWARE: NRF51822 16MHz, 16KB RAM, 256KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (jlink)
PACKAGES:
 - framework-mbed 5.51105.190312 (5.11.5)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
TypeError: string indices must be integers:
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\hansj\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\hansj\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\hansj\.platformio\platforms\nordicnrf51\builder\main.py", line 115:
    target_elf = env.BuildProgram()
  File "C:\Users\hansj\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\hansj\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\hansj\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1041:
    project.search_deps_recursive()
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 404:
    self.process_dependencies()
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 890:
    self.depend_recursive(lb)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 399:
    lb.search_deps_recursive(search_files)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 404:
    self.process_dependencies()
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 283:
    self.depend_recursive(lb)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 399:
    lb.search_deps_recursive(search_files)
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 404:
    self.process_dependencies()
  File "C:\Users\hansj\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 279:
    if item["name"] != lb.name:
=========================== [FAILED] Took 2.53 seconds ===========================
The terminal process terminated with exit code: 1

### Expected Results

Successful compilation

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:bbcmicrobit]
platform = nordicnrf51
board = bbcmicrobit
framework = mbed
lib_deps = microbit-dal
```

**Source file to reproduce issue:**
```cpp
#include <mbed.h>
#include "MicroBitDisplay.h"

int main() {
  while(1) {
  }
}
```

### Additional info

The microbit-dal library hasn't changed. It is still available in the registry. On my machine It is listed as installed in "PIO Home" along with "ble" and "ble-nrf51822"
I get the same behaviour from PIO on macOS (10.15.2)

franzle, 2020-02-25:
I have exactly the same problem with a project with the ble" and "ble-nrf51822" libraries.
I would really need a solution, it stopped compiling with release 4.2.0 onward.
Before it was OK

francesco

knud, 2020-02-26:
Hey,

I wonder if this will help you. I got it working for a Red Bear Nano v1, which uses the nRF51822

https://github.com/knud/RB_Nano_v1

steven