platformio/platform-nordicnrf51 #35: zephir-ble-eddystone example

gerisse, 2020-08-09:
hi,
trying to build the zephir-ble-eddystone example, with bbc microbit board, this issue apparead :

~/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: region `SRAM' overflowed by 572 bytes   (103 %)

Just  a issue with zephyr , or with duo zephyr-pio ?
thanks

valeros, 2020-08-10:
Hi @gerisse ! The issue means that your project doesn't fit into available memory. You can either optimize your SRAM usage or switch to an MCU with larger SRAM size.

closed 2020-08-10 by valeros

gerisse, 2020-08-10:
yes, of course, but i 'm surprised that an standard example given by official source is not running ...

valeros, 2020-08-10:
Could you please try to compile the same example using the native `west` tool?