platformio/platform-nordicnrf51 #37: Error "BLE_API.h: No such file or directory" upon building BLE central test example

vorlovsky, 2020-12-29:
I've imported [BLE_CentralTest](https://github.com/redbear/nRF5x/blob/master/nRF51822/arduino/arduino-1.6.x/hardware/RBL/RBL_nRF51822/libraries/BLE_Examples/examples/BLE_CentralTest/BLE_CentralTest.ino) example from Arduino and tried to build it for nRF51822 with PlatformIO, but got stuck on "BLE_API.h: No such file or directory" error. 
I've tried to select Platformio or Arduino as a library manager upon project import, but no luck. 
Also. there is no such error if to build the same sketch using Arduino IDE. 

What am I doing wrong? How can this dependencies be fixed in PlatformIO?

valeros, 2021-01-04:
Hi @vorlovsky ! The example you mentioned is supposed to work with Arduino core from RedBear. This platform only supports the Arduino nRF5 core https://github.com/sandeepmistry/arduino-nRF5

closed 2021-01-04 by valeros

vorlovsky, 2021-01-05:
@valeros Do you mean I need to use Arduino nRF5 core instead of the one provided by default for the `nordicnrf51` platform? Is it possible to specify it somehow in `platformio.ini`? I see it is possible for STM32 (through `board_build.core` option), but what about nRF5?

valeros, 2021-01-06:
Unfortunately, there is no support for the Arduino core from RedBearLab at the moment.