platformio/platform-nordicnrf51 #39: Issue with missing board ID for RFduino

teknomage, 2022-02-17:
Hello, am attempting to program an RFduino using PlatformIO with ST-link as the programmer. As you may know, RFduino is based on nrf51822, although this particular board has been discontinued by RFdigital. However, I understand that support for RFduino was already added as a board, as documented on this link: https://github.com/platformio/platformio-core/issues/319
However, I am unable to call upon RFduino within PIO on VScode. This is what I've modified inside my project's platformio.ini:

[env:nrf51_dk]
platform = nordicnrf51
framework = arduino
board = rfduino
upload_protocol = stlink
debug_tool = stlink

And when attempting to Build a very simple LED blink example for my RFduino, this is the error I get:

Processing nrf51_dk (platform: nordicnrf51; framework: arduino; board: rfduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'rfduino'
The terminal process "%username%\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Infact, I also checked %username%\.platformio\platforms\nordicnrf51\boards and I do not see RFduino listed in there. I thought this board was already added in as a feature earlier or has it been removed since? Is there some additional board declaration or json file I might be missing??

valeros, 2022-02-17:
Hi @teknomage ! Unfortunately, the support for RFduino was deprecated [several years ago](https://github.com/platformio/platform-nordicnrf51/releases/tag/v4.2.0) and won't be available in the future. You can try to downgrade your platform version to the last one that supported this core, but there is no guarantee it will work:
```ini
[env:nrf51_dk]
platform = nordicnrf51@4.1.0
framework = arduino
board = rfduino
upload_protocol = stlink
debug_tool = stlink
```

closed 2022-02-17 by valeros

teknomage, 2022-02-20:
Thank you for that solution, @valeros! I tried it out with my RFduino and it works well for simple test code like blinking an LED. However, this version fails if I were to attempt to use any of the RFduinoBLE functions that involve the board's radio functions. From what I gather, PIO dropped support of the arduino framework for RFduino from v2.3.0, so I went back to the last supported version and the arduino BLE library functions compile successfully now! Just FYI, this was the change made:

```
[env:rfduino]
platform = nordicnrf51@2.2.0
framework = arduino
board = rfduino
upload_protocol = stlink
debug_tool = stlink
```

However, a new problem crops up when attempting to Upload the build to my RFDuino. In this case, RFDloader is being called instead of OpenOCD (even though the upload_protocol clearly states stlink) and it expects the board to be available on a COM portm presumably via FTDI.  This is the error I'm seeing:

```
Archiving .pio\build\rfduino\libFrameworkArduino.a
Indexing .pio\build\rfduino\libFrameworkArduino.a
Linking .pio\build\rfduino\firmware.elf
Checking size .pio\build\rfduino\firmware.elf
Building .pio\build\rfduino\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.5% (used 452 bytes from 8192 bytes)
Flash: [          ]   2.8% (used 3680 bytes from 131072 bytes)
Configuring upload protocol...
AVAILABLE: stlink
CURRENT: upload_protocol = stlink
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
```
I wish to avoid uploading via FTDI (and in any case it doesnt work for me) because so far, I've only had success with uploading to RFduino via STlink before, so here's my next question: 

Is there a way to enforce uploading the hex through STlink via PIO's OpenOCD, instead of it calling up RFDloader during the Upload process?  Where can I find or modify this setting or command?